avr-llvm/llvm #188: Development still halted?

anonymouse64, 2016-02-14:
In the README, it says that development has halted until avr-llvm has been merged into the llvm trunk. Has this happened? It looks like there are recent commits, and so that would lead me to believe that this merge has happened. 

Just curious...

rjordans, 2016-02-14:
My guess is "yes and no", the merge isn't complete yet so I'm not expecting the main authors of this project to work on many other things while this is happening.  On the other hand, I've become a bit more active and started fixing small things myself, it seems that simple pull requests will still be merged, especially if they fix important things before the merge with llvm trunk is complete.

dylanmckay, 2016-02-14:
Kind of.

I've been working on some things (mostly files that haven't been merged yet).

The backend is about 40% merged so far. See LLVM [trunk](http://llvm.org/viewvc/llvm-project/llvm/trunk/lib/Target/AVR/).

I've come up with a workflow (once something is merged to LLVM trunk, pull changes and resolve merge conflicts with this repo), which makes it easier to add changes. The two repos will be slightly out of sync but so long as each change is integrated into both, it shouldn't matter.

There are some outstanding issues which need to be fixed before the merge is completed (see `blocks-upstreaming` issue label), so it is inevitable that some changes are made.

richardsmithjj, 2016-03-31:
I was checking up on this, your link to the LLVM trunk is returning a 404, any further updates?

galorin, 2016-03-31:
I was looking into doing something where your project came up.  Your link to the LLVM trunk is currently a 404.  Any updates on the status, or is it still hovering at around 40% merged?

dylanmckay, 2016-03-31:
it's probably about 60% merged, with about 20% in one big patch that is awaiting reviewal.

`avr-llvm.org` has also been removed.

dylanmckay, 2016-03-31:
Here are the two patches:

* http://reviews.llvm.org/D16311
* http://reviews.llvm.org/D17906 (big one)

shepmaster, 2016-05-07:
> http://reviews.llvm.org/D17906 (big one)

Merged! 🎉

shepmaster, 2016-05-07:
For those also following along at home, I've been using [Dylan's feed](http://reviews.llvm.org/p/dylanmckay/feed/) to watch all the reviews happen.

eltang, 2016-05-19:
Is there a timeline for the completion of this project? Some C code I'm optimizing has become very fragile, and I suspect that avr-gcc might have a bug.

dylanmckay, 2016-05-19:
@eltang it's currently 1/3rd of the way into being merged into LLVM trunk. I haven't been adding features/fixing bugs during the process.

@shepmaster however is working on getting LLVM to successfully compile Rust's core library and work correctly, and has been doing major work to make it happen.

The truth is though, that there is likely to be many more bugs in AVR-LLVM than in AVR-GCC. I suggest filing a bug report or messaging the [avr-gcc mailing list](mailto:avr-gcc-list@nongnu.org).

eltang, 2016-05-19:
@dylanmckay I'm hesitant to file a bug report because I'm not sure if there is one. I don't know asm; I only know that one version of my code works while another seemingly identical one does not.

dylanmckay, 2016-05-19:
Still no harm in messaging the list.

eltang, 2016-05-20:
@dylanmckay avr-gcc is fine. It turned out that I made an insanely stupid mistake in the code.

shepmaster, 2016-05-20:
> an insanely stupid mistake

Just for my own education, can you share what happened? I'm helping with the Rust fork that supports AVR and I'm curious if Rust could have helped prevent whatever case occurred.

eltang, 2016-05-20:
@shepmaster I substituted a pre-decrement operator with post-decrement operator thinking I was improving the code. I did eventually achieve my aim with a different approach.

closed 2016-09-26 by dylanmckay

eltang, 2016-09-26:
@dylanmckay Has development resumed?

dylanmckay, 2016-09-26:
I'm putting a bunch of work into it now. I'm mostly focusing on the stuff needed for Rust support, but if you're blocked on anything, feel free to mention it :)