avr-llvm/llvm #192: avr-llvm compilation fails on OSX

PeterZegelin, 2016-05-09:
After following all the instructions on the Wiki for building avr-llvm the make process failed about 31% in lib/Target/AVR/AVRFrameLowering.cpp

The specific error is: error: virtual function
      'eliminateCallFramePseudoInstr' has a different return type ('void') than the function it overrides
      (which has return type 'MachineBasicBlock::iterator' (aka
      'MachineInstrBundleIterator<llvm::MachineInstr>'))

I 'fixed' this by specifying the correct return type and returning MBB.erase(MI) and compilation  continued only to fail again shortly after in AVRISelDAGToDAG.cpp with a another function having different return types:
error: virtual function 'Select' has a
      different return type ('llvm::SDNode *') than the function it overrides (which has return type 'void')
  SDNode *Select(SDNode *N) override;

dylanmckay, 2016-05-09:
@shepmaster: This sounds related to your recent change?

@PeterZegelin: If you want, feel free to submit a PR with your fixes :)

shepmaster, 2016-05-09:
I think this is fixed in #191

KarstenB, 2016-05-14:
Hey @shepmaster, I checked out your branch, and it runs into slightly different compilation problem:
[ 74%] Building CXX object tools/clang/lib/Basic/CMakeFiles/clangBasic.dir/Targets.cpp.o
llvm/tools/clang/lib/Basic/Targets.cpp:6855:5: error: use of undeclared identifier 'DataLayoutString'; did you mean
      'DataLayoutStringSI'?
    DataLayoutString = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8";
    ^~~~~~~~~~~~~~~~
    DataLayoutStringSI
llvm/tools/clang/lib/Basic/Targets.cpp:1839:26: note: 'DataLayoutStringSI' declared here
static const char *const DataLayoutStringSI =
                         ^
/Users/karstenbecker/Downloads/llvm/tools/clang/lib/Basic/Targets.cpp:6855:22: error: read-only variable is not assignable
    DataLayoutString = "e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8";
    ~~~~~~~~~~~~~~~~ ^
2 errors generated.

shepmaster, 2016-05-14:
@KarstenB to be sure, you mean [this branch](https://github.com/shepmaster/llvm/tree/track-upstream), right? Which SHA are you at? The branch compiled at last check, as you can see by the PR being green 😸

KarstenB, 2016-05-14:
This happens to me with: 
```
 git log
commit f8c2489edc3d2f7d7da52188b652b12ef35aa71e
Author: Jake Goulding <jake.goulding@gmail.com>
Date:   Thu May 12 12:32:51 2016 -0400

    Add a missing semicolon
```
I essentially did:
```
$ git clone -b track-upstream --depth 1 https://github.com/shepmaster/llvm.git
$ cd llvm/tools/
$ git clone --depth 1 https://github.com/avr-llvm/clang.git
$ cd ../projects/
$ git clone --depth 1 https://github.com/avr-llvm/compiler-rt.git
$ cd ..
$ mkdir build
$ cd build/
$ cmake -G "Unix Makefiles" -DLLVM_TARGETS_TO_BUILD="AVR" ../
$ make
```

shepmaster, 2016-05-14:
Oh, I missed you were compiling clang! For my usecase, I don't need clang, so I haven't been doing anything with that. It's probably likely that you will need to update the clang and compiler-rt repositories to new(er) versions to correspond to the new(er) LLVM versions.

dylanmckay, 2016-05-18:
I'm gonna update clang now.

dylanmckay, 2016-05-18:
Clang now compiles, and is also a part of the Travis-CI build.

closed 2016-05-18 by dylanmckay

KarstenB, 2016-05-19:
Thanks, it now works for me :)