avr-llvm/llvm #200: tinyavr (ATtiny10) code generator incorrectly generates MOVW

cpldcpu, 2016-05-29:
I just noticed, that the code generator generates a word reg-reg move instruction (MOVW) also for the tinyavr core (ATtiny10). This is not support according to the datasheet.

See attached example.

[avrtest.zip](https://github.com/avr-llvm/llvm/files/288596/avrtest.zip)

dylanmckay, 2016-05-30:
I wouldn't be surprised to be honest, I haven't worked on AVR Tiny support very much.

Good thing is though that this is an easy fix :)

@cpldcpu: Is `MOVW` unsupported by all Tiny cores or just the 10?

dylanmckay, 2016-05-30:
Looking at the [datasheet](http://www.atmel.com/images/atmel-2586-avr-8-bit-microcontroller-attiny25-attiny45-attiny85_datasheet.pdf) to the ATtiny25, it looks like it's not all of the Tiny cores.

dylanmckay, 2016-05-30:
Pulling up the [datasheet](http://www.atmel.com/images/atmel-8127-avr-8-bit-microcontroller-attiny4-attiny5-attiny9-attiny10_datasheet.pdf) for the ATtiny10 family, it looks like these are the devices that don't support the instruction.

* ATtiny4
* ATtiny5
* ATtiny9
* ATtiny10

dylanmckay, 2016-05-30:
It looks like in `AVRInstrInfo::copyPhysReg`, we were unconditionally creating `MOVW` instructions, ignoring the subtarget settings.

Am fixing it up now to use two 8-bit `MOV` instructions if `MOVW` is unsupported.

Would you be able to test that the fix works?

dylanmckay, 2016-05-30:
Should be fixed as of f6524cd0. @cpldcpu can you confirm this?

cpldcpu, 2016-05-30:
All devices with the "avrtiny" architecture _do not_ support MOVW: ATtiny 4/5/9/10/20/40/102/104

This is just a small subset of the "ATtiny" series. The ATtiny 25 is based on the AVR25 architecture.

Slightly confusing...

I will test later.

Btw - I came across a huge load of issues in connection with assembler code. Is this currently a focus?

dylanmckay, 2016-05-30:
So, to be clear, we should disable `MOVW` for all tiny cores except the `25`, `45`, and `85`?

> Btw - I came across a huge load of issues in connection with assembler code. Is this currently a focus?

What kind of issues?

cpldcpu, 2016-05-30:
Nono, only for the "tinyavr" / "reduced core" architecture. See Avr-gcc mapping:

http://www.nongnu.org/avr-libc/user-manual/using_tools.html

Regarding assembler: Can provide more bug reports later. I have been trying to compile V-USB.

dylanmckay, 2016-05-30:
The table of subtarget features [we have](https://github.com/avr-llvm/llvm/blob/avr-support/lib/Target/AVR/AVR.td#L263-L505) is pretty much taken from that data, so after f6524cd this should now work correctly for all Tiny devices.

dylanmckay, 2016-05-30:
@cpldcpu if you want to open an issue, go ahead :)

cpldcpu, 2016-05-30:
Might as well add the latest two:

def : Device<"attiny102", FamilyAVRTiny, ELFArchAVRTiny>;
def : Device<"attiny104", FamilyAVRTiny, ELFArchAVRTiny>;

Will open an issue later.

dylanmckay, 2016-05-30:
Have added those devices as of 0c265ab.

cpldcpu, 2016-05-30:
Ok, recompiled and tested my code. Looks looks good now.

closed 2016-05-30 by cpldcpu