avr-llvm/llvm #205: Failed to build

AndreasGe0, 2016-06-02:
Hello everyone, 

I am a new user and I hope this is the correct place for my question. I am trying to build clang and llvm using the directions found here:
http://clang.llvm.org/get_started.html

Everything works fine until it reaches 92% of the build. Then this happens.

[image]

Any ideas??? 
Thank you

shepmaster, 2016-06-02:
`ld terminated with signal 9` is pretty strange. You didn't run out of disk space or anything environmental like that, did you? Anything pertinent in the system log?

dylanmckay, 2016-06-02:
I used to get this when I had 4GB memory and was linking clang and running make in parallel.

The fix is to do less things at once and pass '-j1' to make on one core.

After upgrading to 8GB, it will happily link clang and do other things at the same time.

dylanmckay, 2016-06-02:
To be clear, this is the out of memory killer killing GNU make

AndreasGe0, 2016-06-02:
I am inside a Virtual machine. The operating system is linux ubuntu 16.04 with 100Gb hdd, 4 cpu cores, 4 Gb ram. To be honest I realised that the vm while was working was verry laggy. maybe is the memory thing you mention above? Shall I increase the memory to 8 Gb and try again? 

Ok guys. I am trying again with 8gb of ram! Thank you.

AndreasGe0, 2016-06-05:
Just to know you were correct. Indeed with 8Gb of RAM it worked.

closed 2016-06-05 by AndreasGe0