avr-llvm/llvm #213: Expected either Y or Z register

shepmaster, 2016-06-15:
While compiling the Rust libcore, I get the error

> Expected either Y or Z register

I will update in the next day or two with actual reproduction steps, but figured I'd open it early just in case you magically can figure it out 🌴

shepmaster, 2016-06-17:
```llvm
define void @atomic_store(i16) {
entry-block:
  store atomic i16 %0, i16* undef monotonic, align 2
  ret void
}
```

Compile to object code with `llc -march=avr < y-or-z.ll | llvm-mc -triple avr -mattr=avr6 -filetype=obj`

dylanmckay, 2016-06-19:
I can't seem to reproduce this. I've ran this on the `avr-support` branch _and_ the same branch (but with the data layout string changed to the one you gave in #214).

My command line:

```
./llc -march=avr jake.ll | ./llvm-mc -triple avr -mattr=avr6 -filetype=obj
```

shepmaster, 2016-06-19:
> My command line:

Dumb question, but doesn't `./llc -march=avr jake.ll` not have any standard out as the assembly is written to the file `jake.s`?

Here's the assembly if that helps:

```asm
        .text
        .macosx_version_min 10, 11
        .file   "testcases/y-or-z-small.ll"
        .globl  atomic_store
        .p2align        1
        .type   atomic_store,@function
atomic_store:                           ; @atomic_store
; BB#0:                                 ; %entry-block
        in      r0, 63
        cli
        st      X, r24
        std     X+1, r25
        out     63, r0
        ret
.Lfunc_end0:
        .size   atomic_store, .Lfunc_end0-atomic_store
```

dylanmckay, 2016-06-19:
It looks like the issue is with the `std` instruction - the instruction only supports the `X` and `Y` registers.

GAS has the same error when assembling this code.

dylanmckay, 2016-09-25:
Fixed in 4ba1f0a55ee3a49ecaf3746b7d364ce766a98dc8

closed 2016-09-25 by dylanmckay