avr-llvm/llvm #214: Data layout specification is incompatible with atomic loads and stores

shepmaster, 2016-06-15:
Rust generates atomic loads and stores using the alignment specified by the data layout. However, [it's forbidden](https://github.com/avr-llvm/llvm/blob/2860030806d0fb22831192835e2e257622d73575/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp#L3950-L3951) to load from e.g. a 16-bit value with less than 16-bits of alignment.

This can be fixed by changing the data layout:

```
e-p:16:16:16-i8:8:8-i16:16:16-i32:32:32-i64:64:64-f32:32:32-f64:64:64-n8
```

Although I'm not sure what all potential fallout this might have (#213 maybe?)

rjordans, 2016-06-15:
Do we want/have the floating point specified in the above data layout yet?

shepmaster, 2016-06-15:
@rjordans ¯\_(ツ)_/¯ It's [already there in some form](https://github.com/avr-llvm/llvm/blob/2860030806d0fb22831192835e2e257622d73575/lib/Target/AVR/AVRTargetMachine.cpp#L47).

dylanmckay, 2016-06-16:
This seems like a good idea. Would like to wait until you updated #213 so I can check it fixes it.

shepmaster, 2016-06-16:
> Would like to wait until you updated #213 so I can check it fixes it.

Oh, #213 happens *after* I've made this change locally; I'm worried that this change causes that issue. Still trying to reproduce and reduce that issue :-)

Amanieu, 2016-08-13:
Since avr doesn't support atomic operations you should add "max-atomic-width": 0 to the Rust target specification. This will disable atomic types in libcore.

shepmaster, 2016-08-13:
> Since avr doesn't support atomic operations you should add "max-atomic-width": 0 to the Rust target specification. This will disable atomic types in libcore.

We actually have [some minimal emulation of atomic types](https://github.com/avr-llvm/llvm/pull/211): disable interrupts, perform the operation, re-enable interrupts.

dylanmckay, 2016-09-24:
Fixed in 5f12f20bafa50b178395177cb98aef7de5f68e73

closed 2016-09-25 by dylanmckay