avr-llvm/llvm #217: Upgrade to the 3.9 branch?

shepmaster, 2016-08-02:
Looks like the 3.9 branch has been formed. Rust upstream has upgraded to that version. Any idea if the as-yet-unmerged code here could be rebased on top of that?

dylanmckay, 2016-09-24:
I updated us to LLVM master within the last week

dylanmckay, 2016-09-24:
It looks like I updated _just_ before the 3.9 branch was tagged. I'll update to the exact branch now.

dylanmckay, 2016-09-24:
Updated to the 3.9 branch

closed 2016-09-24 by dylanmckay

shepmaster, 2016-09-24:
Hmm. I'd already updated to 3.9 in 4ab852dc50945acc2f9261d5fd8aa6a2f0a2c920. Wonder what changed.

shepmaster, 2016-09-24:
Oh, perhaps the source of LLVM commits ported to git changed?

shepmaster, 2016-09-24:
It also appears that files like `.travis.yml` have disappeared.

shepmaster, 2016-09-24:
Obviously I'm doing something wrong, I must have pulled the wrong branch. Please disregard my incoherent talking.

shepmaster, 2016-09-24:
Oh. I got bitten by the "main branch is not `master`" issue again. No wonder.

shepmaster, 2016-09-24:
Also, it appears you went to 4.0:

```
$ ./bin/llc --version
LLVM (http://llvm.org/):
  LLVM version 4.0.0
```

dylanmckay, 2016-09-25:
I tried going to 3.9 but it was originally tagged ages ago and we already had newer commits from when I updated last week.

If you need 3.9 specifically, I'm happy to rollback those changes and keep us on that tip.