avr-llvm/llvm #220: llc: Invalid type for expanded integer failed

jackmcu, 2016-09-01:
I'm getting the message: 
 
>llc: /---path---/AVR_LLVM/llvm/lib/CodeGen/SelectionDAG/LegalizeTypes.cpp:824: void llvm::DAGTypeLegalizer::SetExpandedInteger(llvm::SDValue, llvm::SDValue, llvm::SDValue): Assertion `Lo.getValueType() == TLI.getTypeToTransformTo(*DAG.getContext(), Op.getValueType()) && Hi.getValueType() == Lo.getValueType() && "Invalid type for expanded integer"' failed.

**.ll code:**
```
; ModuleID = 'mmodule'

define i16 @main() {
entry:
  call void @pinModeO(i16 6)
  %value = alloca i16
  store i16 0, i16* %value
  %0 = load i16, i16* %value
  %cmptmp = icmp eq i16 %0, 255
  %booltmp = uitofp i1 %cmptmp to double
  %1 = fcmp one double 0.000000e+00, %booltmp
  br i1 %1, label %then, label %ifcont

then:                                             ; preds = %entry
  store i16 0, i16* %value
  br label %ifcont

ifcont:                                           ; preds = %then, %entry
  %2 = load i16, i16* %value
  %cmptmp1 = icmp eq i16 %2, 255
  %booltmp2 = uitofp i1 %cmptmp1 to double
  %3 = fcmp one double 0.000000e+00, %booltmp2
  br i1 %3, label %then3, label %ifcont4

then3:                                            ; preds = %ifcont
  store i16 0, i16* %value
  br label %ifcont4

ifcont4:                                          ; preds = %then3, %ifcont
  ret i16 0
}

declare void @pinModeO(i16)
```

But with this code the error disappear (I just removed the second "if").
The first code is OK when I use x86 llc.

```
; ModuleID = 'aKcent_master_module'

define i16 @main() {
entry:
  call void @pinModeO(i16 6)
  %value = alloca i16
  store i16 0, i16* %value
  %0 = load i16, i16* %value
  %cmptmp = icmp eq i16 %0, 255
  %booltmp = uitofp i1 %cmptmp to double
  %1 = fcmp one double 0.000000e+00, %booltmp
  br i1 %1, label %then, label %ifcont

then:                                             ; preds = %entry
  store i16 0, i16* %value
  br label %ifcont

ifcont:                                           ; preds = %then, %entry
  ret i16 0
}

declare void @pinModeO(i16)
```

Does anybody know what's the problem?

dylanmckay, 2016-09-05:
Would you be able to post the full stacktrace for the assertion error?

dylanmckay, 2016-09-26:
Minimised testcase

```llvm
define void @foo(i16 %a) {
ifcont:
  %cmp_result = icmp eq i16 %a, 255
  %bool_result = uitofp i1 %cmp_result to double
  %result = fcmp one double 0.000000e+00, %bool_result
  br i1 %result, label %then, label %else
then:
  ret void
else:
  ret void
}
```

dylanmckay, 2016-09-26:
Added the testcase in tree

bd648d0c890dfcae4fbf7532d9f6afd9abaaf74b

dylanmckay, 2016-12-07:
Looking at `llc` in a debugger, the operation it is failing on:

```
t26: i64 = select_cc t2, Constant:i16<255>, Constant:i64<4607182418800017408>, Constant:i64<0>, seteq:ch
```

dylanmckay, 2016-12-07:
```
(lldb) print Lo.getValueType()
(llvm::EVT) $2 = {
  V = (SimpleTy = i64)
  LLVMTy = 0x0000000000000000
}
(lldb) print Hi.getValueType()
(llvm::EVT) $3 = {
  V = (SimpleTy = Glue)
  LLVMTy = 0x00000000000000
```

dylanmckay, 2016-12-07:
It looks like we are emitting this out of our custom lowering hook.

dylanmckay, 2016-12-07:
Fixed in [r288905](http://llvm.org/viewvc/llvm-project?view=revision&revision=288905).

closed 2016-12-07 by dylanmckay