avr-llvm/llvm #221: LLVM ERROR: Cannot select: ...

mmavr, 2016-09-04:
The llc gave me the following error:
```
LLVM ERROR: Cannot select: t20: i8,ch = load<LD1[%x], zext from i1> t0, FrameIndex:i16<0>, undef:i16
  t1: i16 = FrameIndex<0>
  t3: i16 = undef
In function: main
```

The `prg.ll` file:
```
; ModuleID = 'new_module'

define i16 @main() {
entry:
  %x = alloca i1
  store i1 true, i1* %x
  %0 = load i1, i1* %x
  %relation_op = icmp eq i1 %0, true
  br i1 %relation_op, label %then, label %else

then:                                             ; preds = %entry
  store i1 false, i1* %x
  br label %ifcont3

else:                                             ; preds = %entry
  %1 = load i1, i1* %x
  %relation_op1 = icmp eq i1 %1, false
  br i1 %relation_op1, label %then2, label %ifcont

then2:                                            ; preds = %else
  store i1 true, i1* %x
  br label %ifcont

ifcont:                                           ; preds = %then2, %else
  br label %ifcont3

ifcont3:                                          ; preds = %ifcont, %then
  ret i16 0
}
```

mmavr, 2016-09-05:
I changed bool type width in my compiler from i1 to i8 (x is bool). That solved my problem.
But I don't know why the llc doesn't work when the bool type width is i1.

dylanmckay, 2016-09-05:
A bunch of the LLVM backends handle `i1` badly (which is pretty sad). This is why almost all frontends define `bool` to be `i8`.

I would definitely like to fix this though. By the looks of this, it is _probably_ failing on the `zext from i1` operation. All that should be needed is to promote the `i1` to an `i8` internally.

dylanmckay, 2016-09-24:
This has been fixed as of 2e0ff8d2168d897362c2222c0bfce76ae5ee23f3.

closed 2016-09-24 by dylanmckay