avr-llvm/llvm #223: error: backend data layout ... does not match expected target description

4ntoine, 2016-09-28:
```
MBA-Anton:bin asmirnov$ ./clang -c -g -Os  -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" ~/Documents/dev/tmp/avr-clang/Blink.cpp -o "/tmp/arduino_test1/Blink.cpp.o" -I/Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include --target=avr
error: backend data layout 'e-p:16:16:16-i8:8:8-i16:16:16-i32:32:32-i64:64:64-f32:32:32-f64:64:64-n8' does not match
      expected target description 'e-p:16:8:8-i8:8:8-i16:8:8-i32:8:8-i64:8:8-f32:8:8-f64:8:8-n8'
1 error generated.
MBA-Anton:bin asmirnov$ ./clang --version
clang version 4.0.0 (https://github.com/avr-llvm/clang.git f9591adbe0f75aa58f733f5c1bb99b9e87302d05) (llvm/llvm.git bd648d0c890dfcae4fbf7532d9f6afd9abaaf74b)
Target: x86_64-apple-darwin15.4.0
Thread model: posix
InstalledDir: /Volumes/Transcend/dev/src/llvm-avr/build/bin/.
MBA-Anton:bin asmirnov$ cat ~/Documents/dev/tmp/avr-clang/Blink.cpp
#line 1 "Blink.ino"
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.
 
  This example code is in the public domain.
 */
 
// Pin 13 has an LED connected on most Arduino boards.
// give it a name:
#include "Arduino.h"
void setup();
void loop();
#line 10
int led = 13;

// walkaround for 'weak referencing' issue
void serialEvent() {

}

// the setup routine runs once when you press reset:
void setup() {                
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);     
}

// TODO : remove
// just for testing until ((weak)) is supported
/*
void serialEvent() {

}
*/

// the loop routine runs over and over again forever:
void loop() {
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);               // wait for a second
}
```

dylanmckay, 2016-09-28:
Probably fallout for the fix from #214. We probably have to copy the data layout string to `clang`.

dylanmckay, 2016-09-28:
Should be fixed in this avr-llvm/clang commit

https://github.com/avr-llvm/clang/commit/c6fdb09fb9bc5cf5c4b52b2d6f39e6349e942f3b

4ntoine, 2016-09-28:
updated clang but now it can't be compiled:
```
[ 85%] Building CXX object tools/clang/lib/AST/CMakeFiles/clangAST.dir/StmtViz.cpp.o
[ 85%] Building CXX object tools/clang/lib/Sema/CMakeFiles/clangSema.dir/SemaTemplateInstantiateDecl.cpp.o
[ 85%] Building CXX object tools/clang/lib/Sema/CMakeFiles/clangSema.dir/SemaTemplateVariadic.cpp.o
[ 85%] Building CXX object tools/clang/lib/CodeGen/CMakeFiles/clangCodeGen.dir/CodeGenFunction.cpp.o
[ 85%] Building CXX object tools/clang/lib/AST/CMakeFiles/clangAST.dir/TemplateBase.cpp.o
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:263:48: error: no type named
      'OptimizationRemark' in namespace 'llvm'
    void OptimizationRemarkHandler(const llvm::OptimizationRemark &D);
                                         ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:264:48: error: no type named
      'OptimizationRemarkMissed' in namespace 'llvm'
    void OptimizationRemarkHandler(const llvm::OptimizationRemarkMissed &D);
                                         ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:265:48: error: no type named
      'OptimizationRemarkAnalysis' in namespace 'llvm'
    void OptimizationRemarkHandler(const llvm::OptimizationRemarkAnalysis &D);
                                         ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:267:21: error: no type named
      'OptimizationRemarkAnalysisFPCommute' in namespace 'llvm'
        const llvm::OptimizationRemarkAnalysisFPCommute &D);
              ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:269:21: error: no type named
      'OptimizationRemarkAnalysisAliasing' in namespace 'llvm'
        const llvm::OptimizationRemarkAnalysisAliasing &D);
              ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:533:17: error: no type named
      'OptimizationRemark' in namespace 'llvm'
    const llvm::OptimizationRemark &D) {
          ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:542:17: error: no type named
      'OptimizationRemarkMissed' in namespace 'llvm'
    const llvm::OptimizationRemarkMissed &D) {
          ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:553:17: error: no type named
      'OptimizationRemarkAnalysis' in namespace 'llvm'
    const llvm::OptimizationRemarkAnalysis &D) {
          ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:566:17: error: no type named
      'OptimizationRemarkAnalysisFPCommute' in namespace 'llvm'
    const llvm::OptimizationRemarkAnalysisFPCommute &D) {
          ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:579:17: error: no type named
      'OptimizationRemarkAnalysisAliasing' in namespace 'llvm'
    const llvm::OptimizationRemarkAnalysisAliasing &D) {
          ~~~~~~^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:623:36: error: use of undeclared
      identifier 'OptimizationRemark'; did you mean 'DK_OptimizationRemark'?
    OptimizationRemarkHandler(cast<OptimizationRemark>(DI));
                                   ^~~~~~~~~~~~~~~~~~
                                   DK_OptimizationRemark
/Volumes/Transcend/dev/src/llvm-avr/llvm/include/llvm/IR/DiagnosticInfo.h:59:3: note: 'DK_OptimizationRemark' declared
      here
  DK_OptimizationRemark,
  ^
/Volumes/Transcend/dev/src/llvm-avr/llvm/tools/clang/lib/CodeGen/CodeGenAction.cpp:623:31: error: no matching function
      for call to 'cast'
    OptimizationRemarkHandler(cast<OptimizationRemark>(DI));
                              ^~~~~~~~~~~~~~~~~~~~~~~~
/Volumes/Transcend/dev/src/llvm-avr/llvm/include/llvm/Support/Casting.h:222:1: note: candidate template ignored:
      invalid explicitly-specified argument for template parameter 'X'
cast(const Y &Val) {
^
/Volumes/Transcend/dev/src/llvm-avr/llvm/include/llvm/Support/Casting.h:229:44: note: candidate template ignored:
      invalid explicitly-specified argument for template parameter 'X'
inline typename cast_retty<X, Y>::ret_type cast(Y &Val) {
                                           ^
/Volumes/Transcend/dev/src/llvm-avr/llvm/include/llvm/Support/Casting.h:236:46: note: candidate template ignored:
      invalid explicitly-specified argument for template parameter 'X'
[... 81 lines cut ...]
                                   DK_OptimizationRemarkAnalysisAliasing
/Volumes/Transcend/dev/src/llvm-avr/llvm/include/llvm/IR/DiagnosticInfo.h:63:3: note: 
      'DK_OptimizationRemarkAnalysisAliasing' declared here
  DK_OptimizationRemarkAnalysisAliasing,
  ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [tools/clang/lib/CodeGen/CMakeFiles/clangCodeGen.dir/CodeGenAction.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 85%] Building CXX object tools/clang/lib/Sema/CMakeFiles/clangSema.dir/SemaType.cpp.o
[ 85%] Building CXX object tools/clang/lib/AST/CMakeFiles/clangAST.dir/TemplateName.cpp.o
[ 85%] Building CXX object tools/clang/lib/AST/CMakeFiles/clangAST.dir/Type.cpp.o
[ 85%] Building CXX object tools/clang/lib/AST/CMakeFiles/clangAST.dir/TypeLoc.cpp.o
[... cut ...]
[ 88%] Building CXX object tools/clang/lib/Sema/CMakeFiles/clangSema.dir/TypeLocBuilder.cpp.o
make[1]: *** [tools/clang/lib/CodeGen/CMakeFiles/clangCodeGen.dir/all] Error 2
[ 88%] Building CXX object tools/clang/lib/AST/CMakeFiles/clangAST.dir/VTableBuilder.cpp.o
[ 88%] Building CXX object tools/clang/lib/AST/CMakeFiles/clangAST.dir/VTTBuilder.cpp.o
[ 88%] Linking CXX static library ../../../../lib/libclangSema.a
[ 88%] Linking CXX static library ../../../../lib/libclangAST.a
[ 88%] Built target clangSema
[ 88%] Built target clangAST
make: *** [all] Error 2

```

4ntoine, 2016-09-28:
```
MBA-Anton:build asmirnov$ cd ../llvm/tools/clang/
MBA-Anton:clang asmirnov$ git log -1
commit c6fdb09fb9bc5cf5c4b52b2d6f39e6349e942f3b
Author: Dylan McKay <dylanmckay34@gmail.com>
Date:   Thu Sep 29 02:29:48 2016 +1300

    [AVR] Update the data layout
    
    We had previously updated the data layout in AVR-LLVM commit
    5f12f20bafa50b178395177cb98aef7de5f68e73.
```

dylanmckay, 2016-09-28:
Seems weird, I'm on the same `clang` commit as you, and my build passes successfully. Perhaps you need to pull changes on AVR-LLVM's `avr-support` branch

4ntoine, 2016-09-28:
it was not clear that i have to update llvm too. after updating i was able to compile avr-llvm/clang and compile blink.cpp file

closed 2016-09-28 by 4ntoine