avr-llvm/llvm #234: AVR target missing in official repo (LLVM_ALL_TARGETS)?

robindegen, 2017-06-30:
I see that AVR llvm is meant to be merged in the llvm upstream, but AVR appears to be missing in LLVM_ALL_TARGETS. Is that on purpose?

dylanmckay, 2017-07-01:
Hey @robindegen 

The AVR backend exists in the LLVM repository but is considered an experimental target, meaning that it is not compiled to default.

In order to compile the backend, pass `-DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=AVR` to `cmake`.

Be aware that the AVR backend does not currently compile (due to breaking API changes in LLVM), I will work on fixing this now, shouldn't take long.

dylanmckay, 2017-07-01:
You can see the current build error via http://lab.llvm.org:8014/builders/llvm-avr-linux/builds/3534.

dylanmckay, 2017-07-01:
I've fixed the error in upstream commit [r306923](http://llvm.org/viewvc/llvm-project?view=revision&revision=306923).

robindegen, 2017-07-03:
@dylanmckay Thanks for your quick response and fix. I'll have a test most likely this afternoon.

closed 2017-08-23 by dylanmckay