avr-llvm/llvm #235: fatal error: error in backend: Cannot select: 0x3b16200: v2i64 = ctlz 0x45c29f0 [ORD=8] [ID=44]

gopinath-r, 2017-09-13:
Hi,
I received the following error while building tensorflow using bazel(using clang)
$ bazel build --crosstool_top=//tools/arm_compiler:toolchain --cpu=armeabi-v7a --verbose_failures --sandbox_debug //tensorflow/tools/pip_package:build_pip_package

fatal error: error in backend: Cannot select: 0x3b16200: v2i64 = ctlz 0x45c29f0 [ORD=8] [ID=44]
  0x45c29f0: v2i64 = bitcast 0x3b16ec0 [ORD=7] [ID=43]
    0x3b16ec0: v4i32 = or 0x3bafaa0, 0x458b200 [ORD=7] [ID=42]
      0x3bafaa0: v4i32 = xor 0x45af7a0, 0x3baf880 [ORD=6] [ID=41]
        0x45af7a0: v4i32 = bitcast 0x45c1b10 [ORD=6] [ID=40]
          0x45c1b10: v2i64 = ARMISD::VSHL 0x3baf990, 0x4543a90 [ORD=4] [ID=38]
            0x3baf990: v2i64 = bitcast 0x45af030 [ORD=3] [ID=32]
              0x45af030: v2f64,ch = load 0x4562170, 0x458aa90, 0x3b16fd0<LD16[%lsr.iv345347](align=8)(tbaa=<0x3b1a498>)> [ORD=3] [ID=28]
                0x458aa90: i32,ch = CopyFromReg 0x4562170, 0x3b16750 [ORD=2] [ID=21]
                  0x3b16750: i32 = Register %vreg15 [ID=1]
                0x3b16fd0: i32 = undef [ID=3]
            0x4543a90: i32 = Constant<1> [ID=15]
        0x3baf880: v4i32 = bitcast 0x3bb0100 [ORD=6] [ID=39]
          0x3bb0100: v2i64 = ARMISD::VSHRs 0x3baf990, 0x4544750 [ORD=5] [ID=37]
            0x3baf990: v2i64 = bitcast 0x45af030 [ORD=3] [ID=32]
              0x45af030: v2f64,ch = load 0x4562170, 0x458aa90, 0x3b16fd0<LD16[%lsr.iv345347](align=8)(tbaa=<0x3b1a498>)> [ORD=3] [ID=28]
                0x458aa90: i32,ch = CopyFromReg 0x4562170, 0x3b16750 [ORD=2] [ID=21]
                  0x3b16750: i32 = Register %vreg15 [ID=1]
                0x3b16fd0: i32 = undef [ID=3]
            0x4544750: i32 = Constant<63> [ID=4]
      0x458b200: v4i32 = bitcast 0x4544310 [ORD=4] [ID=36]
        0x4544310: v2f64,ch = load 0x4562170, 0x426f6d0, 0x3b16fd0<LD16[ConstantPool]> [ORD=4] [ID=31]
          0x426f6d0: i32 = ARMISD::Wrapper 0x3b16530 [ID=27]
            0x3b16530: i32 = TargetConstantPool<<4 x i32> <i32 1, i32 0, i32 1, i32 0>> 0 [ID=20]
          0x3b16fd0: i32 = undef [ID=3]
In function: _ZNK6google8protobuf8internal12ExtensionSet9Extension8ByteSizeEi
clang: error: clang frontend command failed with exit code 70 (use -v to see invocation)
Ubuntu clang version 3.6.2-3ubuntu2 (tags/RELEASE_362/final) (based on LLVM 3.6.2)
Target: armv7a-arm-linux-gnueabif
Thread model: posix
clang: note: diagnostic msg: PLEASE submit a bug report to http://bugs.debian.org/ and include the crash backtrace, preprocessed source, and associated run script.
clang: note: diagnostic msg: 
********************

do anyone have idea about this?

shepmaster, 2017-09-13:
> --cpu=armeabi-v7a 

This repository is for a *fork* of LLVM dedicated to supporting the AVR chip. I think you have found the wrong place to report this. In fact, if you continue reading your pasted message, it tells you where to go:

>  PLEASE submit a bug report to http://bugs.debian.org/ and include the crash backtrace, preprocessed source, and associated run script.

closed 2017-09-13 by shepmaster

dylanmckay, 2017-09-14:
By the way @gopinath-r - this isn't a debian bug, so if this is still an issue in current master (I don't think it is, I remember working on the `ctlz` stuff in the past few months), the place to report it would be https://bugs.llvm.org/